EmbeddedRPC/erpc #326: [BUG] string length not enforced in client or server

taldinj, 2022-11-24:
Hi,
First off, trying out eRPC and impressed by what I see so far. Great work.

I have a command with a maximum of string length associated with the parameter. The max length (@max_length) is not properly enforced in either client nor server:
- On client side the generated code simply writes the string with writeString(strlen(), (const char*)str, (const char*)), no validation is done
- On server side it actually looks worse: 
 * codec->readString() is used, populating size and pointer (uint32_t str_len,  char * str_local)
 * allocation is done with the @max_length+1
 * If the read string is not NULL and allocation was successful memcpy() is used with the length of the read string (possible buffer overflow if string sent exceeds @max_length)

The issues I see:
1. Client side can and should avoid sending content that violets the @max_length
2. No validation of size, possible buffer overflow(?)

Thank you!

Hadatko, 2022-11-24:
HI @taldinj, you may be correct that there is missing validation. Thank you for your input. Once i will have time i will improve this behavior.

amgross, 2022-12-01:
PR #328 is for solving this issue

amgross, 2022-12-01:
As @taldinj mentioned, we have here security issue od buffer overflow. so I will be happy to push this change to 1.10.0

Hadatko, 2022-12-05:
@amgross Thank you for your effort in our project ;)

closed 2023-01-25 by MichalPrincNXP