EmbeddedRPC/erpc #327: [BUG]  Len of none null, but nullable, buffer, len is written twice

amgross, 2022-11-29:
**Describe the bug**

In case of nullable buffer, if the buffer is not null the client writes the length twice. Once explicitly and once as part of writeBinary
**To Reproduce**

run erpcgen on this example file:
[example.erpc.txt](https://github.com/EmbeddedRPC/erpc/files/10110644/example.erpc.txt)
You will see in the output client side:
```
        **codec->write(len);**

        if (buff == NULL)
        {
            codec->writeNullFlag(true);
        }
        else
        {
            codec->writeNullFlag(false);
            **codec->writeBinary(len, buff);**
        }
```
and in server side:
```
   **codec->read(&len);**

    codec->readNullFlag(&isNull);
    if (!isNull)
    {
        uint32_t lengthTemp_0;
        uint8_t * dataTemp_0;
        **codec->readBinary(&lengthTemp_0, &dataTemp_0);**
        **len = lengthTemp_0;**
        if ((lengthTemp_0 <= 10) && (dataTemp_0 != NULL))
        {
            buff = (uint8_t *) erpc_malloc(10 * sizeof(uint8_t));
            if ((buff == NULL) && (lengthTemp_0 > 0))
            {
                codec->updateStatus(kErpcStatus_MemoryError);
            }
            else
            {
                memcpy(buff, dataTemp_0, lengthTemp_0);
            }
        }
        else
        {
            codec->updateStatus(kErpcStatus_Fail);
        }
    }
    else
    {
        buff = NULL;
    }
```
**Expected behavior**

Client side:

        if (buff == NULL)
        {
            codec->writeNullFlag(true);
            **codec->write(len);**
        }
        else
        {
            codec->writeNullFlag(false);
            **codec->writeBinary(len, buff);**
        }
server side:
 ```
codec->readNullFlag(&isNull);
    if (!isNull)
    {
        uint32_t lengthTemp_0;
        uint8_t * dataTemp_0;
        **codec->readBinary(&lengthTemp_0, &dataTemp_0);**
        **len = lengthTemp_0;**
        if ((lengthTemp_0 <= 10) && (dataTemp_0 != NULL))
        {
            buff = (uint8_t *) erpc_malloc(10 * sizeof(uint8_t));
            if ((buff == NULL) && (lengthTemp_0 > 0))
            {
                codec->updateStatus(kErpcStatus_MemoryError);
            }
            else
            {
                memcpy(buff, dataTemp_0, lengthTemp_0);
            }
        }
        else
        {
            codec->updateStatus(kErpcStatus_Fail);
        }
    }
    else
    {
        **codec->read(&len);**
        buff = NULL;
    }
```

**Desktop (please complete the following information):**

- OS:Windows
- eRPC Version:1.9.1

**Steps you didn't forgot to do**

- [X] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

Changing this behavior in just one side, client or server, will break applications.

Hence I think it can't be part of version 1.10.XX
It probably will need to be as part of 2.0.0

Hadatko, 2022-11-29:
Nice catch @amgross , are you willing to create PR for this bug?

amgross, 2022-11-29:
I tried to do it first, but I failed to understand the flow in the templates.

Hadatko, 2022-11-29:
I see what is the issue. The fix has to be done on completely different places. I will provide fix on your branch.

amgross, 2022-11-29:
> I will provide fix on your branch

Which branch? the PR I made is for different issue.

> we can do more complex change next year once i will finish multiple client support

Just reminding my disclaimer at  **Additional context** here, this is breaking change.

>If you see simpler solution let me know

If I will think of such ting I will, currently not.

Hadatko, 2022-12-05:
@amgross Sorry i saw your PR for short time and connect it together.
Definitely it is breaking change as we will need change serialization steps.