EmbeddedRPC/erpc #329: UART or USB_CDC

Second222None, 2022-11-30:
Hello,

I am trying to add **erpc** to my application. My application runs on STM32 with FreeRTOS. Now it contains USB_CDC middleware and exports a Virtual Port to PC. Which API should I use?
- erpc_transport_cmsis_uart_init() **OR**
- erpc_transport_usb_cdc_init()
Does the latest **erpc** support USB_CDC now?

**Steps you didn't forgot to do**

- [ ] I checked if there is no related issue opened/closed.
- [ ] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [ ] I looked in documentation if there is related information.

Hadatko, 2022-12-05:
Based on your output you are missing Linux or NXP SDK dependencies. It means there is no port for STM32 and you should do it on your own if you want it. If you provide PR in that case we will appretiate it.

Depends on which driver you want to use on STM32 (if there is something already existing and can ported here, or you need create also driver by your self.)

Hadatko, 2022-12-05:
fsl_serial is only for Windows, Linux and Apple currently:
[image]

Second222None, 2022-12-06:
> Based on your output you are missing Linux or NXP SDK dependencies. It means there is no port for STM32 and you should do it on your own if you want it. If you provide PR in that case we will [...]

I really love to do this kind of work and create a PR.
What I can think of that needs to be modified is implement **SerialManager_*** kind functions based on STM32 provided APIs.
Please correct me if I am wrong. And hoping for detail instructions.

Hadatko, 2022-12-06:
@Second222None Looking forward for you PR ;)
If you would like to use just simple serial on board with freertos you can take a look to [erpc_uart_cmsis_transport.hpp](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_uart_cmsis_transport.hpp)
If you want to use this CDC feature you need modify header and source file for CDC [erpc_usb_cdc_transport.hpp](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_usb_cdc_transport.hpp) [erpc_usb_cdc_transport.cpp](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_usb_cdc_transport.cpp)

Likely you need modify API for your case. Maybe the port can be inspired by existing thread ports in port directory.
serial files can be removed from your project for STM as you don't need them

Second222None, 2022-12-09:
Hi, 

A lot of code has been written on my server side (STM32 with FreeRTOS). Including：
- Reimplement functions in UsbCdcTransport class
- Create a FreeRTOS task/thread to run erpc server

The server side exposes a Virtual COM port to ubuntu20.04.  Now I'd like to do some tests on matrix-mutiply case and hoping that you could give some advice on the client-side code. The simplest way. I like python :).
Which should I use on the client side:
- erpc_transport_cmsis_uart_init **Or**
- erpc_transport_usb_cdc_init

Thank you!!

Second222None, 2022-12-09:
Thank you! Got it. And sorry for delete my reply as the internet connection is not very well. 
I have solve the problem on client side. I receive the **Resource Busy** Error. Maybe there some issue on my server side. 

```
eRPC request is sent to the server
Traceback (most recent call last):
  File "matrix_multiply.py", line 167, in <module>
    runClient(transport)
  File "matrix_multiply.py", line 121, in runClient
    client.erpcMatrixMultiply(matrix1, matrix2, resultMatrix)
  File "/root/FOC/erpc/examples/matrix_multiply_tcp_python/service/erpc_matrix_multiply/client.py", line 49, in erpcMatrixMultiply
    self._clientManager.perform_request(request)
  File "/usr/local/lib/python3.8/dist-packages/erpc-1.9.1-py3.8.egg/erpc/client.py", line 64, in perform_request
  File "/usr/local/lib/python3.8/dist-packages/erpc-1.9.1-py3.8.egg/erpc/transport.py", line 78, in send
  File "/usr/local/lib/python3.8/dist-packages/erpc-1.9.1-py3.8.egg/erpc/transport.py", line 117, in _base_send
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 986, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 229, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/usb/core.py", line 178, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 829, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/local/lib/python3.8/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
```

Hadatko, 2022-12-09:
@Second222None i cannot help. Try solution here: https://stackoverflow.com/questions/67453535/i-keep-getting-usb-core-usberror-errno-16-resource-busy-when-trying-to-read