EmbeddedRPC/erpc #331: [QUESTION] Codec/FramedTransport for header inside message buffer

matth314, 2022-12-12:
Hi,

I was implementing a uart driver for a low power MCU, and I had a hard time with the way the current `FramedTransport` work, aka sending the header first and the message buffer second. I simplified my efforts by putting the framed header inside the message buffer in the `BasicCodec`, for the `FramedTransport` to be able to send everything in one go. 
Would that be something you'd be interested in ?

I created my own classes for my `Codec` and `FramedTransport` implementations, but I could also write that inside the BasicCodec and FramedTransport classes, either with `#defines` or with options selectable at runtime.

Regards

**Steps you didn't forgot to do**

- [ ] I checked if there is no related issue opened/closed.
- [ ] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [ ] I looked in documentation if there is related information.

Hadatko, 2022-12-12:
I think better is to have your codec and framedTransport version.

amgross, 2022-12-12:
> Hi @matth314 I think better is to have your codec and framedTransport version. Looking forward for your PR as i think more people were interested into this feature in past.

Indeed #284

Hadatko, 2023-08-10:
+1
Today i remembered you. Current implementation is not working with rpmsg tty transport as it is expecting all data at once. Which is i think bug in rpmsg-lite, but it is working with python serial as python serial is sending all at once. So i will try solve both cases

FYI: https://github.com/EmbeddedRPC/erpc/issues/372

Hadatko, 2023-09-19:
@MichalPrincNXP this is not yet solved. RPMSG tty is not solving this issue. Myabe it was closed automatically so i reopened it.

closed 2023-10-24 by MichalPrincNXP