EmbeddedRPC/erpc #333: [QUESTION] Don't understand line use

amgross, 2022-12-13:
**If you didn't find answer in existing open/closed issues you may ask here**

Why this line is needed?
https://github.com/EmbeddedRPC/erpc/blob/89f7d9b2eaa3a30742bcd9ed82d815f39173af29/erpcgen/src/templates/c_server_source.template#L274

**Steps you didn't forgot to do**

- [ ] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [ ] I looked in documentation if there is related information.

Hadatko, 2022-12-14:
In release target, for static allocation policy the service variable is not used. In that case analysing tools may, compilers,... may complain that variable is unused. For that case you will do that line above to let them know that you know and you want keep it.

amgross, 2022-12-14:
Got you, thanks :)

closed 2022-12-14 by amgross