EmbeddedRPC/erpc #336: ERPC_TYPE__DEFINITIONS when multiple distinct clients are used

Diagano, 2023-01-10:
Hi,

WIill try to keep short and to the pont as this is not that big of a deal, here is our use casse:
We have multiple groups working of the same platform but on different aspects of our platform. We do not want nor do we Intersect with the other clients which expose completly different set of functionalities to other cores on seperate transports and managed by different groups.
Now what I am concerned about is the fact that common header content is protected from redefinition by #ifndef/#define ERPC_TYPE_DEFINITIONS. Which will conflict if any code attempts to use more than once of these ERPC apps.
I found [this discussion](https://github.com/EmbeddedRPC/erpc/pull/181) but I don't think its relevant to my case.

Is there any wat to control this other than the obvious manual modification or tinkering with the generator code?
If not I would lilke to suggest considering doing so at least in the appplication level, that is for top level IDL

progarme **<program_name>**

the define would be:

ERPC_**<program_name>**_TYPE_DEFINITIONS.

Thank you

Hadatko, 2023-01-10:
Hi, sounds good to me.

Hadatko, 2023-01-10:
Hi @Diagano , i made changes based on your requests. Until the changes will be merged to master you can take changes to your fork and compile erpcgen on your own and use it. 

In 1-2weeks there should be also other PR merged, which will build binaries for Linux and Mac within every PR and make it available for all users.

Diagano, 2023-01-10:
That was quick :)
Thank you, will have a look and will replace my own simplified local changes (that are likely not properly covering all cases)

Hadatko, 2023-01-10:
Happy to help ;)

Hadatko, 2023-01-12:
Well i am about to delete this option anyway due to develop on other issue. The header file will be always created.

Diagano, 2023-01-12:
> Well i am about to delete this option anyway due to develop on other issue. The header file will be always created.

Maybe this merits a seperate Q :)
I liked the fact that the common types do not repeat in each header when using @types_header. So after you delete this option a common header file will always be created + the specific headers will still hold the common types under ERPC_TYPE_DEFINITIONS ?

Hadatko, 2023-01-12:
Common declarations will be only in one file.

closed 2023-03-07 by MichalPrincNXP