EmbeddedRPC/erpc #339: erpcgen compilation when both py2 as system default while py3 is present

Diagano, 2023-01-12:
Hi,

We only recently moved to 1.9.1 and see that the default is now to use py3.
In our basic docker env we have both setup and have 2.7 as the default (using update-alternatives)
Its very simple to overcome but more to the fact that out of the box without any change (external or in paths.mk) the erpcgen compilation fails even though python3 is installed and accessible as python3.
I do see that if "python" was not linked use of "python3" is attempted, but not if python linked and is different version the make will fail. Maybe there is a better way at it than the current one in paths.mk?

Maybe even reverse the current from:

PYTHON ?= python
ifeq (, $(shell which $(PYTHON)))
    PYTHON= python3
endif

to:

PYTHON ?= python3
ifeq (, $(shell which $(PYTHON)))
    PYTHON= python
endif

Hadatko, 2023-01-12:
Hi, could you wait until end of this week? New eRPC will be released which should fix your issue. It will print python2.7, but it will use python3 as default. Also we are providing prebuilt applications here + once new PR will be merge we will provide also application per PR merged.

 I think it is inside this commit https://github.com/EmbeddedRPC/erpc/commit/c9e1caaa16893d1a17697c794a61fc9ad5c527d8

Diagano, 2023-01-12:
No problem waiting with it. I looked at the change and is solve the issue for us.
Altough am wondering if the printout is not missleading and if the reverse I have suggested will possibly clear that as well.

Hadatko, 2023-01-13:
Well maybe for most cases but not for all. My system doesn't have python3 only python which is version 3. I will try to improve makefile to do not print version. It is bothering me too (on other machines).

Diagano, 2023-01-16:
> Well maybe for most cases but not for all. My system doesn't have python3 only python which is version 3. I will try to improve makefile to do not print version. It is bothering me too (on other machines).

Not sure I understood your point. Reversing the value assignment as I suggested (at least over the original 1.9.1 make) does not prevent from setteling into "python" if "python3" does not exist.
So all that is done here below (vs 1.9.1) is replace the default assignment from ="python" to ="python3" and viceversa in the ifeq (in Bold).

**PYTHON ?= python3**
ifeq (, $(shell which $(PYTHON)))
    **PYTHON= python**
endif

This will try use "python3" and settle for "python" otherwise.
The later lines _without any change_ will still validate that the selection exists and the version of that selection.

ifeq (, $(shell which $(PYTHON)))
    $(error "No python found")
endif
ifneq (3, $(shell $(PYTHON) --version | cut -c 8-8))
    $(error "Please install and use Python3")
endif

So this minor change covers my issue ("python" exists and defaults to py2) while python3 exists as well. It will also cover your system config since it will fail to find "python3" and will fallback to "python" and pass the later validations. Last, in both cases it will only print the selected varient that is to be used.
The only other potential issue I can think off is different varients of py3 pointed to by python and python3, don't know if this is a relevant cnofiguration to consider.

Hadatko, 2023-01-16:
It make sense i think, but i will think about it more.