EmbeddedRPC/erpc #340: transport_arbiter addPendingClient with dynamic allocation policy

Diagano, 2023-01-19:
Hi,

First thanks for the latest drop of 1.10.0.
I have 2 questions please,
The first is more to make sure I understand the "behavior" of TransportArbitrator and the second one is about a potential bug

1. As far as I can see, at least with dynamic allocation policy in use, calls to TransportArbitrator::addPendingClient() will always allow the pending list to grow, as one pending client is allocated at a time and if we "fall" over the edge m_clientFreeList == NULL will result in a new\additional allocation that is added at the head of the client list.

TransportArbitrator::removePendingClient() only moves a used entry to the free list. So in theory each usage "peak" will result in a memory allocation that will not be freed until the arbitrated client containing it is destroyed, is this correct?

2. TransportArbitrator::addPendingClient()

```
    if (m_clientFreeList == NULL)
    {
        info = createPendingClient();
    }
```
createPendingClient() use of ERPC_CREATE_NEW_OBJECT, which in the case of dynamic allocation uses new (std::nothrow), can return NULL which is not tested. The calling code does test for toke=0 (TransportArbitrator::prepareClientReceive) but before that the following part in addPendingClient () will fail:
```
    // Add to active list.
    info->m_next = m_clientList; //<--
    m_clientList = info;
```
Is this analysis correct?

Thanks

Hadatko, 2023-01-24:
Hi, yes it looks like there is missing check for null.

Hadatko, 2023-01-24:
Thank you for your insights ;)

closed 2023-01-26 by MichalPrincNXP