EmbeddedRPC/erpc #342: compile error with Mac Xcode 14

ftab, 2023-01-25:
This is with eRPC 1.10.0

I'm using a MacBook Air M1 (Apple Silicon), so the homebrew paths are different:

```
BISON_ROOT=/opt/homebrew/opt/bison FLEX_ROOT=/opt/homebrew/opt/flex BOOST_ROOT=/opt/homebrew/Cellar/boost/1.81.0_1 make -j
```

The error:

```
/opt/homebrew/Cellar/boost/1.81.0_1/include/boost/lexical_cast/detail/converter_lexical_streams.hpp:310:21: error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Werror,-Wdeprecated-declarations]
```

The workaround:

```diff
diff --git a/mk/flags.mk b/mk/flags.mk
index a42233d..3919b8e 100644
--- a/mk/flags.mk
+++ b/mk/flags.mk
@@ -50,6 +50,8 @@ ifeq "$(is_darwin)" "1"
     LDFLAGS += -L $(BOOST_ROOT)/lib
     LDFLAGS += -L$(FLEX_ROOT)/lib
     INCLUDES += $(FLEX_ROOT)/include
+    # don't warn about sprintf deprecation, see https://github.com/boostorg/boost/issues/688
+    CXXFLAGS += -Wno-deprecated-declarations
 endif
 
 # Need to pass the -U option to GNU ar to turn off deterministic mode, or
```

Until Boost reworks that library to no longer use sprintf, we have to add that flag.

Hadatko, 2023-01-25:
Hi @ftab , could you print this in your terminal? I am curious what it will return: 
```sh
$(and $(findstring Darwin,$(os_name)),1)
```

Hadatko, 2023-01-25:
So far we can fix this: 
[image]

Hadatko, 2023-01-26:
Guys from boost said that it will be fixed in next release. So i would like to not to add this workround flag. Maybe keep it as known issue with how to fix it info.

Hadatko, 2023-01-26:
Maybe we can parse boost version and if it is smaller than 1.82 then apply macro.

ftab, 2023-01-30:
> Hi @ftab , could you print this in your terminal? I am curious what it will return:
> 
> ```shell
> uname -s
> ```
> 
> Also i would like to know your gcc and g++ version which i used to compile code...

```
➜  ~ gcc --version
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: arm64-apple-darwin22.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
➜  ~ g++ --version
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: arm64-apple-darwin22.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
➜  ~ uname -s
Darwin
```

Good to know Boost has a fix planned.

Hadatko, 2023-02-08:
I am about to remove boost dependency. Then this issue should be solved.

closed 2023-02-09 by MichalPrincNXP