EmbeddedRPC/erpc #348: crc16 is using internal crc with 32 bit size

amgross, 2023-02-16:
https://github.com/EmbeddedRPC/erpc/blob/60a3cc1f84351cc3bf1af719d662e979a8f19b1c/erpc_c/infra/erpc_crc16.cpp#L32-L38
Maybe I am missing something.
Why can't the variables `crc`, `temp`, `andValue `, `xorValue `be uint32_t and `byte `be uint8_t ?
It makes things cleaner and I don't see it effects the algorithm

Hadatko, 2023-02-16:
Hi @amgross , so only change you are suggesting is change byte to uint8_t correct?
I am not against it but we need be carefull. I think "byte<<8" can do undefined behavior. Compiler can throw away one byte. So we need recast it then ((uint16_t)(byte)) << 8

amgross, 2023-02-16:
Oops, I wrote wrong, I mean:
Why can't the variables `crc`, `temp`, `andValue `, `xorValue `be **uint16_t**

amgross, 2023-02-16:
About byte<<8,
Please see https://stackoverflow.com/questions/69143163/c-uses-different-data-type-for-arithmetic-in-the-middle-of-an-expression .
`byte<<8` is equivalent to `((unsigned int)byte)<<8`