EmbeddedRPC/erpc #349: [BUG] Compilation issues with ERPC_ALLOCATION_POLICY_STATIC

Diagano, 2023-02-23:
Hi, 

1. Compilation errors on erpc_arbitrated_client_deinit()\erpc_client_deinit()\erpc_server_deinit(). All 3 have simular failure cause

[image]

2. Generated file that has a structure with a string field. The read function fails compilatoin because of call to erpc_malloc(). That is since the include of "erpc_port.h" is under #if ERPC_ALLOCATION_POLICY == ERPC_ALLOCATION_POLICY_DYNAMIC
[image]
[image]
Which was generated from the following struct (This from the IDL) that is later used as input argument:
[image]

**To Reproduce**

#define ERPC_ALLOCATION_POLICY (ERPC_ALLOCATION_POLICY_STATIC)

Is static allocation policy part of the testing?

Hadatko, 2023-03-07:
Hi, we are truly missing Static allocation tests. Static allocation case can be used within some basic types in IDL (int, uint mostly).  I think this may be our next big challenge after multiple client support. But not sure why arbitrator code is not able to compile. I will check and fix.

closed 2023-03-07 by MichalPrincNXP

Diagano, 2023-03-08:
Please note that I gave two issues here, the second one was not addressed by the PR that was closed.
The second issue stems from this template line in c_server_source.template:
```
#if ERPC_ALLOCATION_POLICY == ERPC_ALLOCATION_POLICY_DYNAMIC
#include <new>
#include "erpc_port.h"
#endif
```
This is wrong because even in static policy a struct read\write, for example,  would use a call to **erpc_malloc** if it holds string or binary content as I have shown in our case. So the generated code excluds the include even though there is use of "erpc_malloc" leads to compilation error.
In my case for now I just manually moved the `#include "erpc_port.h"` out of the ifdef.

Hadatko, 2023-03-08:
@Diagano Second case i exaplained. It is not bug, more it is not implemented feature.

Diagano, 2023-03-09:
Sorry, I did not see any comment on the second issue. 
So I am missing something, you may not want to but the generator does produce code that uses it. Is there any PR that changed that?
We work with POLICY_STATIC , the generated code I get with for the use of this kind of struct in the ERPC:
```
struct key_attributes_t
{
    string              key_name;
    key_type_e       key_type;
    uint16 key_size;
}
```
Results in this in the generated server size struct read function:
`data->key_name = (char*) erpc_malloc((key_name_len + 1) * sizeof(char));`

But actually that was only first one. Looking at our generated code I see that in server code for every string\binary input\output.
For example
`GetVersion(in uint32 version_str_max_length, out string version_str @max_length(version_str_max_length)) -> RetCode_e`

has this in the server generated code:
`version_str = (char *) erpc_malloc((version_str_max_length + 1) * sizeof(char));`