EmbeddedRPC/erpc #356: [QUESTION] Bare Metal usage : new used in STATIC_ALLOCATION

NeroSC, 2023-04-20:
I want to use eRPC on a bare metal platform that has no OS, no C library. Goal is also to not do any dynamic allocation. 

I took the matrix example and here is my main file :

```c
#include "erpc_matrix_multiply.h"
#include "erpc_matrix_multiply_server.h"

#include "erpc_transport_setup.h"
#include "erpc_server_setup.h"

#include "header.h"
#include <stdio.h>

void erpcMatrixMultiply(Matrix matrix1, Matrix matrix2, Matrix result_matrix)
{
    /* code for multiplication of matrices */
    
    printf("salut\n");
}

int main(int argc, char *argv[]) {
    const char *host = "localhost";
    const int port = 10000;
    bool isServer = true;
    /* Init eRPC server environment */
    /* TCP transport layer initialization */

    erpc_transport_t transport = erpc_transport_tcp_init(host, port, isServer);
    printf("TCPTransport initialized\n");
    
    /* MessageBufferFactory initialization */
    erpc_mbf_t message_buffer_factory = erpc_mbf_dynamic_init();
    printf("MBF initialized\n");

    /* eRPC server side initialization */
    erpc_server_t server = erpc_server_init(transport, message_buffer_factory);
    printf("Server initialized\n");
    /* connect generated service into server, look into erpc_matrix_multiply_server.h */
    erpc_service_t service = create_MatrixMultiplyService_service();
    printf("Service created\n");

    erpc_add_service_to_server(server, service);
    printf("Service added to Server\n");

    /* run server */
    erpc_server_run(server); /* or erpc_server_poll(); */
    printf("Server created on \"%s:%d\" \n",host, port);
    return 0;
}
```

It seems that the ERPC_ALLOCATION_POLICY checks prevents use of "new" in the following code but its not actually the case : 

```c
erpc_service_t create_MatrixMultiplyService_service(void)
{
    erpc_service_t service;

#if ERPC_ALLOCATION_POLICY == ERPC_ALLOCATION_POLICY_DYNAMIC
    service = new (nothrow) MatrixMultiplyService_service();
#else
    if (s_MatrixMultiplyService_service.isUsed())
    {
        service = NULL;
    }
    else
    {
        s_MatrixMultiplyService_service.construct();
        service = s_MatrixMultiplyService_service.get();
    }
#endif

    return service;
}
```

Indeed, in the `else`, the `s_MatrixMultiplyService_service.construct()` calls the following: 
```c
void construct(void)
    {
        destroy();
        new (m_storage) T; // use of new which is basically dynamic allocation ... 
        m_isConstructed = true;
    }
```
Using gdb this `new` points in the c++ new lib file 
So it doesn't seem to be doing any static allocation in the end

Is there anything I am missing here ? 

- [x] I checked if there is no related issue opened/closed.
- [x] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [x] I looked in documentation if there is related information.

Hadatko, 2023-04-20:
Hello. Static allocation is not fully supported (only for simple messages.) But in this case it is static allocation. There is parameter for new where you are providing memory where object is allocated. The parameter is created in global scope (m_storage).

Hadatko, 2023-06-06:
For example usage of numbers are ok.
Then depends if you have client and server in c or only client....