EmbeddedRPC/erpc #357: [QUESTION] User data accessible in function implementation

snalvc, 2023-04-20:
Hi

I'm writing to inquire if eRPC already supports adding user data to the service creation and invocation process. Specifically, I would like to be able to register a void* pointer when creating a service, and then have that pointer passed to the RPC server function implementation in the service. Just like one is able to pass a void* ptr when creating threads/tasks in pthread/FreeRTOS.

**Steps you didn't forgot to do**

- [x] I checked if there is no related issue opened/closed.
- [x] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [x] I looked in documentation if there is related information.

amgross, 2023-04-20:
Hi @snalvc ,
I don't sure I understand you,
If you want some callback, you can see at #313 
If you want just to send pointer from client to server, as the server and client may be in different memory spaces, address of one of them is not relevant to the other

amgross, 2023-04-20:
So he want that the caller side API and the callee side will be different? I prefer the way it is today. keep it simple

Hadatko, 2023-04-20:
> So he want that the caller side API and the callee side will be different? I prefer the way it is today. keep it simple

I think so.

amgross, 2023-04-23:
I would expect such change will effect more than the IDL