EmbeddedRPC/erpc #358: Why is the response to a client request being processed by the server?

rabryn, 2023-04-21:
Hello,

I've implemented a client and server on a microcontroller as well as a Qt desktop application.  I can successfully call functions in the micro from the Qt application.  However, when I call a function on the Qt application from the microcontroller, the response from the Qt application goes to the server of on the microcontroller.  Here's how I'm declaring the client and server on the microcontroller...

```
/// Thread handle the ERPC server is run on.
static os_thread_t s_thread_handle;
/// Transport.
static erpc::Kw41SerialTransport s_transport;
/// Transport arbitrator.
static erpc::TransportArbitrator s_arbitrator;
/// Codec factory.
static erpc::BasicCodecFactory s_basic_codec_factory;
/// Message buffer factory.
static erpc::Kw41MessageBufferFactory s_msg_buff_factory;
/// CRC16.
static erpc::Crc16 s_crc16;
/// Arbitrated client manager.
static erpc::ArbitratedClientManager s_client;
/// Server.
static erpc::SimpleServer s_server;
/// Global client.
extern erpc::ClientManager *g_client;
```
And here's how I'm configuring the client and server...
```
    s_transport.init();

    s_arbitrator.setSharedTransport( &s_transport );
    s_arbitrator.setCodec( s_basic_codec_factory.create() );
    s_arbitrator.setCrc16( &s_crc16 );

    s_client.setArbitrator( &s_arbitrator );
    s_client.setCodecFactory( &s_basic_codec_factory );
    s_client.setMessageBufferFactory( &s_msg_buff_factory );

    s_server.setTransport( &s_transport );
    s_server.setCodecFactory( &s_basic_codec_factory );
    s_server.setMessageBufferFactory( &s_msg_buff_factory );

    s_server.addService( new dev_led_service() );
    s_server.addService( new dev_str_service() );
    s_server.addService( new dev_timer_service() );

    s_client.setServer( &s_server );
    s_client.setServerThreadId( reinterpret_cast<erpc::Thread::thread_id_t>( s_thread_handle ) );

    g_client = &s_client;

    s_server.run();
```
Here's the IDL for the function in question...
```
//
// This interface defines timer functions that the device can call on the host.
//
@group("host")
interface host_timer
{
    TimerExpired() -> void

} // inteface host_timer
```
When I call `TimerExpired()` on the micro the request is sent to and processed by the Qt application, however, the response sent from the Qt application back to the micro is getting processed by the server.  Specifically, I'm getting this error in `erpc_server.cpp`...
```
...
erpc_status_t Server::processMessage(Codec *codec, message_type_t msgType, uint32_t serviceId, uint32_t methodId,
                                     uint32_t sequence)
{
    erpc_status_t err = kErpcStatus_Success;
    Service *service;

    if ((msgType != kInvocationMessage) && (msgType != kOnewayMessage))
    {
        err = kErpcStatus_InvalidArgument;
    }
...
```
The `msgType` is `kReplyMessage` which fails the check above and breaks out of the server.  I'm curious as to why the micro's client isn't waiting for the response from the Qt application?  It seems as if the micro's client is sending the request and moving on which causes the server to process the response.

I should also mention the micro's request is being made on the application's "main thread" (not the same thread the server is running on).  I'm using FreeRTOS and working on the `develop` branch with a git hash of `e8e2e633f23948291b2191a93a13927cd160843d`.

Finally, thank you for the excellent project!

Ryan

rabryn, 2023-04-22:
I figured out my issue.  I needed to better manage how I was executing the server on a thread.

amgross, 2023-04-23:
@rabryn if you found the root cause (and as I understand it is issue in your setup), can you please close the issue?

closed 2023-04-25 by rabryn