EmbeddedRPC/erpc #374: [QUESTION] Using Arbitrated client with empty server for multithreaded client calls

taldinj, 2023-08-18:
Hi,

Our use case is clinet->server only. So we have used the basic erpc_client_init().
As far as I can see this interface is not thread safe In terms if send receive synchronization (correct?)
I do see that ArbitratedClientManager() solves this with the pending client list . On the other hand ArbitratedClient seems intended for use with Server\Client sharing a transport and not for the our use case.
To force usage of arbitrated client I guess that would force me to create an DispatchServer and run a server thread soley for this  Before I dive into this any furtert, I would like to seek advice:
Is using ArbitratedClient \server without any actual service (possible?) an overkill ?

Thanks

Hadatko, 2023-08-19:
Hi @taldinj, if i understand corectly you want let several threads call client rpc functions. I see some ideas to improve current solution to match your requests, but can be tricky. Anyway currently you would have to setup arbitrator client and server on one side and server on other side. You don't need add services to server (i think so).

Hadatko, 2023-08-19:
Actually i think you can simplyfy first side with using only arbitrator client (no server) and call arbitratedTransport->receive(MessageBuffer) in loop. 
This is interesting and we can make more generic easy use abstraction for your use case and also fill it into documentation.

taldinj, 2023-08-26:
Hi,
Thank you for your replies. Sorry for my late response I was on an off the grip vacation for a long time.
Just to clerify you last suggestion, " using only arbitrator client (no server) and call arbitratedTransport->receive(MessageBuffer) in loop.", that would still have to be done with a thread that keeps this receive loop alive right?

Hadatko, 2023-08-26:
No problem. Yes you are right. It has to be in its own thread. Yes your code snippet looks correct to me ;)