EmbeddedRPC/erpc #375: [FEATURE] Put eRPC C++/Python version into conan build system

Hadatko, 2023-08-25:
**Is your feature request related to a problem? Please describe.**

Recently we started to use conan for downloading and building projects. Maybe we can consider to do support for that. User can then simply download C++/Python code without git, with requested version (similar to pypi which is for python only).

**Describe the solution you'd like**

If we would have some github automatization which would put new version of erpc to conan would be great.
https://conan.io/center/recipes
You can choose c++ topics or directly search for some lib. Unfortunatelly many projects doesn't set topics so c++ topic show only few project. But definitelly there is much more of them.

**Describe alternatives you've considered**

For c++ we don't have such a feature.

**Steps you didn't forgot to do**

- [X] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

**Additional context**

Hadatko, 2023-08-25:
It looks like github has many option for automization: 
[image]

burgerTo, 2025-03-05:
Hello @Hadatko, have there been any attempts to make eRpc available as a conan package? We use conan in our projects and it would be very helpful to include erpc in this way. 

Best regards
Toni

Hadatko, 2025-03-06:
I dont' think this was implemented so far.