EmbeddedRPC/erpc #380: [QUESTION] Can this run on microcontroller such as Raspberry Pi Pico?

cvele1995, 2023-09-01:
Can EmbeddedRPC be used on Raspberry Pi Pico for example?
I have their SDK for development applications, now I want to create RPC server on it and trigger it via UART.
RPC server runs on Raspberry Pi Pico, client runs on Linux PC. Microcontroller and PC are connected via UART.

If this is can be done on Raspberry Pi Pico, is there some guide how to do it, not for Raspberry Pi Pico directly, but any generic microcontroller?

Hadatko, 2023-09-02:
Hello @cvele1995 , if we are talking about Linux OS with Linux serial and pthreads support, then i don't see any issue why it should not work on rpi pico. 

Documentation is here: https://github.com/EmbeddedRPC/erpc/wiki/Getting-Started

As an example can serve tests (e.g. initialization files in test/common folder) or nxp examples for nxp devices where you need register and then you can download examples for some board.

mindforger, 2023-09-02:
The pico is an RP2040 microcontroller, it does not run on linux. But the controller is spreading like a wildfire :)

Hadatko, 2023-09-02:
So then depends, if it is running OS which is not supported, you need implement serial/uart and thread. It should be just these two files.

mindforger, 2023-09-02:
Wonder if i could make the py version run with circuitpython, threading might be an issue since it only provides hardcoded asyncio not threading itself afaik

Hadatko, 2023-09-02:
We actually didn't specify language and i was thinking that you mean C/C++. In case of Python maybe you need just implement new transport if current doesn't fit.

cvele1995, 2023-09-02:
@Hadatko Thanks for responding.
Reason why I mentioned Raspberry Pi Pico was that I wanted to make a test program for EmbeddedRPC to see how it works on other simpler platform. Make a simple program.

I have NXP board and in the past I used mcux-sdk for developing firmware on it.
Now I am using some other product made by NXP that is not mcux-sdk (but it has a similar role).
For UART communication with the CPU I am using LPUART.

When I was looking at the board examples of Embedded RPC, for example [matrix multiply](https://github.com/nxp-mcuxpresso/mcux-sdk-examples/tree/main/frdmk22f/multiprocessor_examples/erpc_server_matrix_multiply_uart)
I noticed it uses [_ARM_DRIVER_USART ](https://github.com/nxp-mcuxpresso/rt-thread/blob/e4b70a10dd4036a1373426be1871a7b33b73d0d9/bsp/lpc54608-LPCXpresso/SDK_2.2_LPCXpresso54608/CMSIS/Driver/Include/Driver_USART.h) structure, meaning it uses CMSIS USART driver.

Can Embedded RPC work with LPUART?
If it can, what would be the steps to integrate it?

Sorry if I make some stupid question, I am new to this.

Regards

Hadatko, 2023-09-02:
HI, no problem. I think we used to have LPuart and uart implementation. But to make implementation easier we swtiched to use cmsis uart which in case of NXP could be used with both lpuart and uart device.
CMSIS: https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_uart_cmsis_transport.hpp

cvele1995, 2023-09-02:
This means I would have to implement my own transport class using LPUART, by inheriting FramedTransport class?

Hadatko, 2023-09-03:
@MichalPrincNXP Does cmsis uart works with nxp lpuart driver or only with full uart driver?

cvele1995, 2023-09-04:
@Hadatko 
When I look at the [UartTransport](https://github.com/EmbeddedRPC/erpc/blob/develop/erpc_c/transports/erpc_uart_cmsis_transport.hpp) class and its implementation, it inherits "FramedTransport" interface.

Is it mandatory for functions "underlyingReceive" and "underlyingSend" to block until next frame comes?

MichalPrincNXP, 2023-09-08:
Hi @cvele1995 , cmsis uart driver is generic API interface that can be used with different underlying uart drivers, incl. lpuart. If you have the sdk package for you nxp board, I would recommend to look at boards/<board name>/cmsis_driver_examples  folder to see how it is used in projects. 
I would also recommend to download the sdk package for either evkmimxrt1170 or lpcxpresso55s36 or frdmk32l2b board that contain erpc_server_matrix_multiply_uart multiprocessor example showing the erpc communication between the PC running python and MCU running C via the uart transport. CMSIS UART transport is used there but different underlying uart drivers are utilized, based on the particular MCU uart peripheral.

Hadatko, 2023-09-27:
I think thisproblematic was explained.

closed 2023-09-27 by Hadatko