EmbeddedRPC/erpc #398: Multiple client support usage

tmordeko, 2024-01-18:
Hi,

Is there an example of usage for the upcoming multiple client support ?

Thanks

Hadatko, 2024-01-18:
Just this so far and tests:
https://github.com/EmbeddedRPC/erpc/pull/392/files

tmordeko, 2024-01-24:
Hi @Hadatko 
Not sure I understand the usage from the test you pointed me to in your previous reply, is it available only over TCP transport  erpc_transport_tcp_init ? does it continue to use same simple server data structure ?

Thanks

Hadatko, 2024-01-25:
Any transport supported transport can be used. Not supported transport can be added trough PR.

amgross, 2024-02-20:
Hi @Hadatko ,
I am probably missing something, where in this example I can find two clients? I see just one.

Hadatko, 2024-02-20:
Hi @amgross, there is no examples with two clients directly. Just example how to init client with new code. If you init two clients and everything in that init is doubled it should work.

amgross, 2024-02-20:
OK, thanks, understand.
This new feature is about one application having two clients talking to two different servers and not one server serving two clients, am I correct?
Is there is plan to support one server serving two clients in parallel?

Hadatko, 2024-02-20:
Yes you are correct. One server is currently serving request on one transport layer, so i would said that depends on transport layer.

closed 2025-01-05 by tmordeko