EmbeddedRPC/erpc #399: Porting to IAR C

darwyy, 2024-01-22:
Hi Guys,
I want to compile the server code in IAR for my cortex M33. Is there any porting guide to build in C? 
Also can I generate C code instead of cpp using erpcgen tool?

Thanks,
David

Hadatko, 2024-01-22:
Hi @darwyy https://github.com/EmbeddedRPC/erpc/wiki/Porting-Guide
Official doc is for previous version. For current version released recently it is currently on branches. But it si similar allthought initialization parameters were changed.

joelguittet, 2024-06-13:
Hello,
I have a similar need, i guess the question of @darwyy was to get PURE C, not some binding around the C++ code.
It seems it's not possible by the way according to the help of the erpcgen tool ?
Any idea of the difficulty to have pure C support instead of C++ ? Anybody already worked on this ?
Joel

Hadatko, 2024-09-25:
Hi, i think so far there is no plan to create pure C implementation. I don't even see reason for that as in C++ we are not using things like STL, ...