EmbeddedRPC/erpc #400: How to generate only C code for erpc file

pioneerAlone, 2024-01-25:
**I am using erpc 1.12 version, but when using erpcgen to generate code, no matter whether "-g c" is specified or not, it always generates the code as shown below. I want to know what codes I need if I only use C, or all of them It is a must, I checked the guide on the wiki, they seem to have not been updated for a long time, which is not very friendly to newbies**

Hadatko, 2024-01-25:
Hi @pioneerAlone , all these files are necessary for running C erpc functions. Implementation is in C++. C generated files contains  C wrapper functions which are calling C++ functions inside.

pioneerAlone, 2024-01-25:
Got ot, 
Is there an example of compiling demo? I tried to compile them but got the following error.
```
g++ -Wall -I. -I/usr/local/include/erpc -L/usr/local/lib c_hello_server.cpp hello_server.cpp  -lerpc -lpthread -o server
g++ -Wall -I. -I/usr/local/include/erpc -L/usr/local/lib c_hello_client.cpp hello_client.cpp -lerpc -lpthread -o client

```
`
q@q-OptiPlex-5050:~/bakewell/gitprj/hello_erpc$ ./build.sh
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x24): undefined reference to `main'
/usr/bin/ld: /tmp/ccp55hAR.o: in function `Hello_server::~Hello_server()':
c_hello_server.cpp:(.text._ZN12Hello_serverD2Ev[_ZN12Hello_serverD5Ev]+0x26): undefined reference to `erpcShim::Hello_interface::~Hello_interface()'
/usr/bin/ld: /tmp/ccp55hAR.o: in function `Hello_server::sayHello(binary_t const*)':
c_hello_server.cpp:(.text._ZN12Hello_server8sayHelloEPK8binary_t[_ZN12Hello_server8sayHelloEPK8binary_t]+0x24): undefined reference to `sayHello'
/usr/bin/ld: /tmp/ccp55hAR.o: in function `erpcShim::Hello_interface::Hello_interface()':
c_hello_server.cpp:(.text._ZN8erpcShim15Hello_interfaceC2Ev[_ZN8erpcShim15Hello_interfaceC5Ev]+0xf): undefined reference to `vtable for erpcShim::Hello_interface'
/usr/bin/ld: /tmp/ccp55hAR.o:(.data.rel.ro._ZTI12Hello_server[_ZTI12Hello_server]+0x10): undefined reference to `typeinfo for erpcShim::Hello_interface'
/usr/bin/ld: /tmp/cc6z8YJQ.o: in function `erpcShim::Hello_service::sayHello_shim(erpc::Codec*, erpc::MessageBufferFactory*, erpc::Transport*, unsigned int)':
hello_server.cpp:(.text+0x2e2): undefined reference to `nestingDetection'
/usr/bin/ld: hello_server.cpp:(.text+0x313): undefined reference to `nestingDetection'
collect2: error: ld returned 1 exit status
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x24): undefined reference to `main'
/usr/bin/ld: /tmp/ccFGWu3n.o: in function `erpcShim::Hello_client::~Hello_client()':
hello_client.cpp:(.text+0x188): undefined reference to `erpcShim::Hello_interface::~Hello_interface()'
/usr/bin/ld: /tmp/ccFGWu3n.o: in function `erpcShim::Hello_interface::Hello_interface()':
hello_client.cpp:(.text._ZN8erpcShim15Hello_interfaceC2Ev[_ZN8erpcShim15Hello_interfaceC5Ev]+0xf): undefined reference to `vtable for erpcShim::Hello_interface'
/usr/bin/ld: /tmp/ccFGWu3n.o:(.data.rel.ro._ZTIN8erpcShim12Hello_clientE[_ZTIN8erpcShim12Hello_clientE]+0x10): undefined reference to `typeinfo for erpcShim::Hello_interface'
collect2: error: ld returned 1 exit status
`

Hadatko, 2024-01-25:
See my comments here: https://github.com/EmbeddedRPC/erpc/issues/399

pioneerAlone, 2024-01-25:
Hello ，

Thanks，
I am trying hello_world example on #398 , but I can't resolve the compilation issue below:

`error: /usr/bin/ld: build/.objs/main_server/linux/x86_64/release/shim/c/hello_world_server.cpp.o: in function `erpcShim::TextService_service::printText_shim(erpc::Codec*, erpc::MessageBufferFactory*, erpc::Transport*, unsigned int)':
hello_world_server.cpp:(.text+0x1e6): undefined reference to `nestingDetection'
/usr/bin/ld: hello_world_server.cpp:(.text+0x213): undefined reference to `nestingDetection'
collect2: error: ld returned 1 exit status
`

Hadatko, 2024-01-25:
I see, it is possible there is an issue within configuration
try disable nested calls detection. Or enable if it is already disabled:
#define ERPC_NESTED_CALLS_DETECTION (ERPC_NESTED_CALLS_DETECTION_DISABLED)

pioneerAlone, 2024-01-25:
I resolved it with the code below, It seems that I should disable or enable when compiling the library?

```
#define ERPC_NESTED_CALLS_DETECTION 1

#if ERPC_NESTED_CALLS_DETECTION
bool nestingDetection;
#endif
```