EmbeddedRPC/erpc #401: Erpc: support for different versions of the messages [QUESTION]

jozefkralovic, 2024-01-25:
Hi all,

Is there implemented ERPC support for message which have 2 different versions. Lets say original message version have 2 parameters. And there will new version of the message with 3 parameters. New parameter is not affecting  functionality of the original 2 parameters.

Are 2 ERPC entities still able to communicate also in case when both entities support same type of the message but in different versions. Lets say message with 2 parameters is sent and receiver (2nd ERPC entity) is able to decode it and provide for other handling also these 2 parameters? 

If not yet supported is it planned in any release plan?

Note: it would be helpful in case when there is not possibility to update firmware at both devices in the same time and new message parameter is not affecting original parameters in the same type of the message.

Thank you for help

**Steps you didn't forgot to do**

- [ x ] I checked if there is no related issue opened/closed.
- [ x ] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [ x ] I looked in documentation if there is related information.

Hadatko, 2024-01-25:
Hi i am not right now sure. But maybe it is possible when the third command is IN type. 
Theoretically we can support that in the future (something similar to grpc maybe)

jozefkralovic, 2024-01-30:
Thank you for your response.

Is there used any known coding scheme (coding/decoding)? Something as Type–length–value coding scheme. Or any own algorithms?

Thank you for help