EmbeddedRPC/erpc #409: [BUG] Zephyr UART transport does not works with CDC ACM transport

gabbla, 2024-02-22:
## Describe the bug

Zephyr RTOS provides a transparent API to interact with physical UARTs and USB CDC ACM ones, so in theory one could create an `UartTransport` with either one.

Using a physical UART with the mentioned transport works fine, however, when switching to a CDC ACM one, the server always returns 16 (`kErpcStatus_ReceiveFailed`) after receiving an RPC call.
The issue resides in the `UartTransport::underlyingReceive` function: the logic handles the case in which the `uart_receive_buf` is not yet filled with the needed bytes, so it waits on a semaphore and only after the get the `erpcStatus` is marked as success. However, when the `uart_receive_buf` already has the needed bytes, the `erpcStatus` is not updated and the read wrongfully fails.

https://github.com/EmbeddedRPC/erpc/blob/85d3dd8656ccce4c2d929a69484fb8d88ec6c6c3/erpc_c/transports/erpc_uart_zephyr_transport.cpp#L154-L182

I believe this may fail in other context even with physical UARTs, but I was not able to validate the such statement.

I'm about to open a PR with a proposed solution.

## To Reproduce

Simply create an `erpc::UartTransport` using an USB CDC ACM node instead of a physical UART.

## Expected behavior

The UartTransport should works with any kind of UART.

## Desktop

The test were conducted with eRPC Version v1.12.0 

## Steps you didn't forgot to do

- [x] I checked if there is no related issue opened/closed.
- [x] I checked that there doesn't exist opened PR which is solving this issue.

Hadatko, 2024-02-22:
Hi @gabbla , thank you for your inputs. I am don't have skills with Zephyr, but i think your requested changes are not good. If i understand you correctly you mean that buffer in second case contains header+body. In first case header and body comes as two messages correct?
If i am correct for CDC you would need similar implementation that we have for erpc_rpmsg_tty_rtos_transport.cpp. That implementation count that it can receive two messages one header one body, or it can receive all at once. 
In current implementation the header buffer has strict length and you cannot fill it with more data. But if you override another function instead, you will be able to get all data at once.

gabbla, 2024-02-22:
Hi @Hadatko, thank you for the quick replay. Yes, while using CDC the buffer is filled immediately with header + body, meanwhile using the UART the header and body came byte by byte (on my platform). Platform with bigger UART FIFO (in hardware) may rise an interrupt when they have more than a single byte ready to be read (similar to what is happening with CDC).

My change should cover all cases, because if the condition
https://github.com/EmbeddedRPC/erpc/blob/85d3dd8656ccce4c2d929a69484fb8d88ec6c6c3/erpc_c/transports/erpc_uart_zephyr_transport.cpp#L158
is false, that means we already have all the bytes we need, otherwise we need to wait. So it should not matter if we waited the semaphore or we already had all the bytes, `erpcStatus` should be `kErpcStatus_Success` in both cases. Or am I missing come cases?

I do not understand what do you mean with
> In current implementation the header buffer has strict length and you cannot fill it with more data. But if you override another function instead, you will be able to get all data at once.

Can you please elaborate?

Gab

JanKomarekNXP, 2024-03-04:
Hi @gabbla, thank you for your contribution. Upon reviewing the bug, I think that the pull request from @gabbla is fine. In the interrupt, I am using the [`uart_fifo_read()`](https://docs.zephyrproject.org/apidoc/latest/group__uart__interrupt.html#gab10942076ac47ecff29e924098049398) to load everything in the `uart_receive_buf` buffer during the interupt. Enough data may be already loaded before calling `UartTransport::underlyingReceive`, which may prevent erpcStatus from changing to kErpcStatus_Success. I have already marged the PR to the develop branch.

closed 2024-03-04 by JanKomarekNXP