EmbeddedRPC/erpc #418: [BUG] Using multiple eRPC clients/server in a C++ project lead to redefinition of struct binary_t

PhilippHaefele, 2024-03-26:
## Describe the bug
We're using multiple eRPC files to generate different server / client connections in one project.  
We're using the new namespace feature of 1.12.0 and assign each eRPC file it's own namespace.

As long as we do not have conflicting type names, we do not have any problems.  
But as `binary_t` is implicitly generated for `binary`, we can't solve this issue by our self.

## To Reproduce
Create two `.erpc` files with both containing a function that uses `binary` and a different `@namespace(".....")`.  
Then include both server (and or clients, so we get the "...common.hpp" file of both included) in on C++ file.

## Expected behavior
The code generator, generates code without type conflicts / no compile errors

I think it should be easy by adding the namespace also to the `...._common.hpp` file, while generating it via the code generator.  

## Desktop (please complete the following information)

- OS: n.a.
- eRPC Version: [1.12.0](https://github.com/EmbeddedRPC/erpc/releases/tag/1.12.0)

PhilippHaefele, 2024-03-27:
Just for other having this issue and can't wait for the fix:  
I did try some workarounds (in C++17) and found a ugly one

You need to include (at least) the  common header in your desired namespace before including others (this means that you need to be very careful with include order).

````cpp
namespace YourNamespace
{
#include "...._common.hpp"
}

#include "...._client.hpp"
````

When fixing this, it should be considered how to handle the constants, as for C usage they need to be global (no namespace). In C++ this extern constants (which are to for some reason weak?) should better be namespaced constexpr, so they can be used in the function prototypes / interface definition e.g. for array sizes

rbrune-polaris, 2024-10-23:
I'm facing this issue as well, and can't find a workaround.  The namespace thing I'm not sure I understand.  I currently have the client and server has part of the same component, so it would be in the same namespace.  I am switching from an older implementation of eRPC which didn't have this issue.