EmbeddedRPC/erpc #426: [QUESTION]

akpour, 2024-06-03:
ERPC forward and backward compatibility

- Can someone comment about ERPC backward and forward compatibility scenarios?

1- what if arguments are added/removed from the interfaces signature
2- what if argument types are added/removed
3- what if newer APIs are added, etc

Has backward and forward compatibility being properly thought about in ERPC?

amgross, 2024-06-03:
Hi
From what I know:
Client shouldn't call server if their api are different, calling API that not exist in the server will cause server side to 'crash' ithout returning error to the client, calling API that changed most of the time will cause server crash but may lead also to calling the API with wrong parameters.
But if some of the API are equal and some are different (due to API addition or change or whatever) still the equal API can be called, if the enum that representing them have same values in both sides.
In general of course it is good practice not to use client and server from different generation, and if erpcgen executable/erpc code itself are different between server and client it should be checked carefully that none breaking change added to erpc between the versions.

amgross, 2024-07-07:
@akpour if this answered you please close the issue