EmbeddedRPC/erpc #429: [QUESTION] dose this project still maintained

amgross, 2024-07-07:
Hi @Hadatko @MichalPrincNXP , dose this repository is still maintained?

For a long time there wasn't change and for longer there werent answers to questions.

- [x] I checked if there is no related issue opened/closed.
- [x] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [x] I looked in documentation if there is related information.

MichalPrincNXP, 2024-07-23:
Hello @amgross , the new release has been done recently: https://github.com/EmbeddedRPC/erpc/releases/tag/1.13.0
The project is in maintenance mode now due to limited capacities, will try to go through open issues and PR soon ...
Regards
Michal

amgross, 2024-07-24:
OK, thank you very much for this great repo (:

closed 2024-07-24 by amgross