EmbeddedRPC/erpc #433: [FEATURE]I hope to release a library based on the C language. C++ libraries are very unfriendly to embedded sdk environments.

ueJone, 2024-08-27:
## Is your feature request related to a problem? Please describe

## Describe the solution you'd like

## Describe alternatives you've considered

## Steps you didn't forgot to do

- [ ] I checked if there is no related issue opened/closed.
- [ ] I checked that there doesn't exist opened PR which is solving this issue.

## Additional context

ueJone, 2024-08-28:
> Hi @ueJone , i don't think there is estimation for rewriting C++ code to C. But you can use C wrapper functions to call API directly from C source code. Many embedded libraries works like that.

Thank you for your reply. I will try this method. Thank you again

ueJone, 2024-08-29:
> Hi @ueJone , i don't think there is estimation for rewriting C++ code to C. But you can use C wrapper functions to call API directly from C source code. Many embedded libraries works like that.

The IDE I am using is MDK-Arm. I test other C wrapper functions and they worked properly, but there will be many compilation errors when I compile the ERPC library.
[image]

It seems like we need to open C++11. so I need to switch the compiler from AC5 to AC6. In this case, I will need to spend a long time adapting the project to AC6.

amgross, 2024-09-05:
Are you sure you are compiling erpc_client_manager.cpp here with cpp compiler and not c compiler?

ueJone, 2024-09-06:
> Are you sure you are compiling erpc_client_manager.cpp here with cpp compiler and not c compiler?

The compiler is ARM Compiler integrated by MDK 5.39.(As shown in the figure).I have successfully compiled demo of C++ with the same configuration, so I think it could support compiling the code of C++. And the demo of C++ will be shown below.

[image]

```c

// robot.cpp

#include <iostream>

#include "robot.h"

void Robot::sayHi()
{
    std::cout << "Hi, I am " << name_ << "!\n";
}

```

```c

// robot.h

#pragma once

#include <string>

class Robot
{
public:
    Robot(std::string name) : name_(name) {}

    void sayHi();

private:
    std::string name_;
};

```

```c
// robot_c_api.cpp

#include "robot_c_api.h"
#include "robot.h"

#ifdef __cplusplus
extern "C" {
#endif

void Robot_sayHi(const char *name)
{
    Robot robot(name);
    robot.sayHi();
}

#ifdef __cplusplus
}
#endif

```

```c

// robot_c_api.h

#pragma once

#ifdef __cplusplus
extern "C" {
#endif

void Robot_sayHi(const char *name);

#ifdef __cplusplus
}
#endif
```

amgross, 2024-09-08:
Hi @ueJone 
According [ARM support](https://developer.arm.com/documentation/ka002877/latest/) You should be able use c11 by:
```
Arm Compiler 5.05, supplied with the current version of Keil MDK, supports most of the C\+\+11 compiler features.

To enable the C\+\+ language extensions, enter in the µVision IDE under Project - Options - C/C\+\+ - Misc Controls: --cpp11.
```

ueJone, 2024-09-11:
> Hi @ueJone According [ARM support](https://developer.arm.com/documentation/ka002877/latest/) You should be able use c11 by:
>
> ```
> Arm Compiler 5.05, supplied with the current version of Keil MDK, [...]

After enabling cpp11, there are no more errors mentioned above. But the original SDK code will have many conflicts related to C++. Because I have few interfaces for RPC, so I plan to implement it myself using C language.
Thanks.

ueJone, 2024-09-11:
Hi @amgross 
I compiled erpc into a static library with the cpp switch. But the API generated by IDL files is also in C++, so I still need to open cpp to compile my code with the API. Therefore, a conflict arose.

Now I have a question: Compile the API interface files into static libraries first and can I call the API in my own code  without the cpp switch?

## It seems that the conflicting are from the standard library
[image]

## There will also be many grammar errors at the same time
[image]

Hadatko, 2024-09-11:
Hi, as i mentioned before, you can generate C api. Only implementation is in c++.

ueJone, 2024-09-11:
> Hi, as i mentioned before, you can generate C api. Only implementation is in c++.

Is that right? First, use the C++compiler to compile ERPC and  the APIs generated by generpc into static libraries, and then compile my code with the both libraries by the C compiler.

amgross, 2024-09-12:
I also using pure c code with erpc library compiled in cpp.
I hope you mean to link both libraries with c linker and not compile (as you can't compile cpp code with c compiler), and yes, I think it should work.the
About the errors, the conflicts errors seems to me like you are trying to link two different standard libraries together, I suggest you to check in ARM documentation how to correctly compile link CPP and C libraries. the drv_gpio errors are weird to me, as those are compiler errors and you claim you compile the c code with your regular c compiler, so how it related to the cpp code?

Hadatko, 2024-09-12:
@ueJone It is possible for you to export minimal project -> Publicly available sdk+init code+erpc call - without application logic. Something like minimal publicable project export so we can try to modify build to make it working?

ueJone, 2024-09-12:
> @ueJone It is possible for you to export minimal project -> Publicly available sdk+init code+erpc call - without application logic. Something like minimal publicable project export so we can try to modify build to make it working?

project path: bsp/hc32f4a0_app/project.uvprojx

IDL file:
```c
/*!
 * Copyright (c) 2024, Z.
 */

@output_dir("erpc_python")

program myRPC

interface Binary {
    oneway sendBinary(binary a)
}

```

SDK: 
[rt-thread-4.1.0.zip](https://github.com/user-attachments/files/16977493/rt-thread-4.1.0.zip)

ueJone, 2024-09-13:
> I also using pure c code with erpc library compiled in cpp. I hope you mean to link both libraries with c linker and not compile (as you can't compile cpp code with c compiler), and yes, I think it [...]

I compiled the erpc_c and API files separately into static libraries, then add them into my project, compiled and ran them successfully. 

Thranks!

However, the api files generated by generpc must also be compiled into a library to use, which is a bit too tedious

Hadatko, 2024-09-13:
I think the limitation you are facing is because you are using old compiler type and you should switch to the newer one:
https://developer.arm.com/documentation/kan298/latest/

amgross, 2024-09-13:
Indeed the .h files that auto generated are the API and needed to be used also outside the erpc library.
Hence they are .h and not .hpp so it will also compile with c code and not just cpp.
If everything now working, pleas cose the issue.

ueJone, 2024-09-13:
> Indeed the .h files that auto generated are the API and needed to be used also outside the erpc library. Hence they are .h and not .hpp so it will also compile with c code and not just cpp. If everything now working, pleas cose the issue.

Yes, I think so. It is best if the api file that auto generated can be used with the c compiler. 

I want to wait a few days before closing this issue, beacuse  I'm looking forward to the test results of @Hadatko 
Thanks!

Hadatko, 2024-09-23:
Soryy busy week. I installed  IDE so far. Next week i should have plenty of time

Hadatko, 2024-10-27:
I didn't have. Sorry again. Too many projects. But i think the only way would be upgrading compiler as in link above.