EmbeddedRPC/erpc #436: [BUG] When enabling ERPC_MESSAGE_LOGGING, results in compilation failure

rbrune-polaris, 2024-10-25:
## Describe the bug
I attempted to enable ERPC_MESSAGE_LOGGING in hopes to isolate what wasn't working.  When I went to compile, however, I would get compilation errors, such as:

...erpc_c/infra/erpc_client_manager.cpp:98:59: error: cannot convert 'erpc::MessageBuffer' to 'erpc::MessageBuffer*'
|    98 |             err = logMessage(request.getCodec()->getBuffer());
|       |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
|       |                                                           |
|       |                                                           erpc::MessageBuffer

It appears the logMessage() accepts a pointer, but the getBuffer() returns a direct MessageBuffer.

## To Reproduce

## Expected behavior

Enable ERPC_MESSAGE_LOGGING in the erpc_config.h file.

## Screenshots

## Desktop (please complete the following information)

- OS: Building for FreeRTOS on Linux
- eRPC Version: 1.13.0

## Steps you didn't forgot to do

- [x] I checked if there is no related issue opened/closed.
- [X] I checked that there doesn't exist opened PR which is solving this issue.

## Additional context

Hadatko, 2024-10-27:
Hi @rbrune-polaris , i see. This part of code was not tested. Could you provide fix? I guess you need change line to provide address to  the logMessage function.
something like:  `err = logMessage(&request.getCodec()->getBuffer());`

rbrune-polaris, 2024-10-28:
> Hi @rbrune-polaris , i see. This part of code was not tested. Could you provide fix? I guess you need change line to provide address to the logMessage function. something like: `err = logMessage(&(request.getCodec()->getBuffer()));`

I can look into it if I get a moment.  You're right that the fix should be easy...just have to change every instance of this.

rbrune-polaris, 2024-12-11:
Turns out the fix isn't quite so straightforward.  There are places where the compiler won't allow just passing the address of the return value of getBuffer(), so it involves more than  such a simple change.  I would need to dig into it more when I have more time.

djmuhlestein, 2025-04-10:
A simple fix is to pull the MessageBuffer out of the parameter call
```
auto buf = request.getCodec()->getBuffer();
logMessage(&buf);
```
I'd prefer personally a more elegant solution for passing a ptr around but this fixes the compiler errors for me without having to add -fpermissive to the cpp_flags.