EmbeddedRPC/erpc #442: [QUESTION]chould i use erpc to transmit some simple audio?

kabutosama, 2024-12-20:
I need to use the host computer to transmit some simple audio back and forth to the ESP32S3, but I'm not sure how to proceed.

Hadatko, 2024-12-20:
Hello. If there are different types of messages, you need create IDL file -> generate C++ code -> Write transport layer for ESP (e.g. update tcp transport with ESP tcp library) -> init and call RPC functions.

Hadatko, 2024-12-20:
FYI Funny is that i think this is first ESP question here and i made PR into ESP few hours ago :DDD (not eRPC related https://github.com/esphome/esphome/pull/7988)

kabutosama, 2024-12-20:
Thank you for your response. However, I am using a CDC serial port, and I would like to know if eRPC is suitable for this type of stream-based transmission.annnnnnd 
That's quite a coincidence! It seems like we are working on similar projects related to this.

Hadatko, 2024-12-20:
It is possible that you will need implement esp port files (mutex, threads, ...) and transport related files (serial send/receive, init). I am not sure but if it is single threaded then you can use bm port.

kabutosama, 2024-12-20:
thank you ,now I finished my transport ,but should i use array? list ????? ，oneway？？？？？，in my opnion，erpc is a call but stream-based transmission，it is strange 。。。。

Hadatko, 2024-12-20:
List is dynamic sized message. Array is with fixed size. That depends on your needs. Oneway is sending messages to other side asynchronously (it is not waiting for respond).

kabutosama, 2024-12-20:
thank you  ,thank you ,i decide use array

amgross, 2025-01-02:
@kabutosama can you please close the issue if you got the needed answer?

closed 2025-01-06 by kabutosama