EmbeddedRPC/erpc #450: [QUESTION] About  mem for rpmsg transport

ADragonKnight, 2025-02-17:
hi nxp
  erpc is a great project for rpc. I use rpmsg as transport for imx8mp to make Core-A to Core-M and Core-M to Core-A.
but i can't understand in erpc_c/transports/erpc_rpmsg_lite_transport.cpp function RPMsgTransport::send
````c
erpc_status_t RPMsgTransport::send(MessageBuffer *message)
{
    int32_t ret_val = rpmsg_lite_send_nocopy(s_rpmsg, m_rpmsg_ept, m_dst_addr, reinterpret_cast<char *>(message->get()),
                                             message->getUsed());
    message->set(NULL, 0);

    return (ret_val != RL_SUCCESS) ? kErpcStatus_SendFailed : kErpcStatus_Success;
}
````

if there make message buf to NULL and size to zero, how to free this message buf? I know this memory from virtio queue. Could it be that don't need free? I test this problem. add log to setup_mbf_rpmsg and log format is
````c
printf("create: %p %d\n", buf, size);

printf("dispose: %p %d\n", tmp, buf->ghetLength());
````
and test result is 
````shell
dispose: 0x80e65664 16
create: 0x55408010 4080
dispose: 0x80e65664 16
create: 0x55409010 4080
dispose: 0x80e65664 16
create: 0x5540a010  4080
````
memory address for 0x55408010  0x55409010 0x5540a010 don't free.  look forward for your reply.

Hadatko, 2025-02-17:
Hi @ADragonKnight , are you aware of that message buffers created on one side are freed on other side? There are N/2 buffer amounts for every side. Each half is used for creating and writing on one side and reading and freeing on other side.

ADragonKnight, 2025-02-17:
Wow，I know this from OpenAMP project. this framework very efficient。 thinks for your reply.