EmbeddedRPC/erpc #457: [QUESTION]TCP transport with ThreadX

jacqueshsu, 2025-06-05:
## If you didn't find answer in existing open/closed issues you may ask here

## Steps you didn't forgot to do

- [ ] I checked if there is no related issue opened/closed.
- [ ] I checked that there doesn't exist opened/closed PR which is solving this issue.
- [ ] I looked in documentation if there is related information.

amgross, 2025-06-06:
In some erpc_threading_XXX.cpp ports the init function is optional and in other ports (like threadx) it is mandatory.
I don't sure if it is a valid issue because I think the TCP functions used in this transport are not part of the standard libraries (for sure they are not in threadx) hence I don't sure we really need to support it. What do you think @MichalPrincNXP ?

jacqueshsu, 2025-06-06:
Do you have any example code for threadx? I check the history of the issues, it seems the threadx has been verified with uart interface.

amgross, 2025-06-08:
I don't have